monkeyboard/Wiegand-Protocol-Library-for-Arduino #3: Library is not really interrupt-save

Apollon77, 2015-06-26:
Hi,

I'm currently started coding for Arduino and learned some things about interrupts and the pitfalls and some things I do not find in your library:

1.) All variables that are shared between "Interrupt service routines" (ReadD0/1) should be declared as "volatile"
2.) When using the values in the main-loop you should disable interrupts because else an interrupt can interrupt the code and can bring you in potential race conditions

When my fingerprint-scanner is available then I will try to send in a pull request (because else I can not test the changes).

jpliew, 2015-06-26:
Thanks mate,

1) Yes, you are right. 

2) The Wiegand pulses D1 and D0 are not overlapping, even if it is overlapping, the ISR need to be reentrant. So we do not need to disable interrupts when inside the ISR.

Cheers
JP

Apollon77, 2015-06-26:
2) I did not mean inside the ISR (there are interrupts disabled by default). I especially meaned in "doWiegandConversion" ... so the code outside of the ISR that uses the variables that are also used in the ISRs.

jpliew, 2015-06-26:
2) Yes, in the main-loop, there is a potential where the interrupt will change the variables. I should have just read the variables and store and use separate variables for the main loop.

Apollon77, 2015-06-26:
See my pull request #4 :-)
It is not tested (i only know it compiles without an error) ... but I would exactly change this for the interrupt things ...

closed 2015-08-02 by jpliew