monkeyboard/Wiegand-Protocol-Library-for-Arduino #9: no response - SR-2400 RFID Reader

eoinverling, 2015-08-25:
Hi,

I have:-

* Arduino UNO
* SR-2400 RFID reader
[image]

http://www.awid.com/index.php?option=com_content&view=article&id=536:sr-2400&catid=978:readers-proximity-lf&Itemid=448

running the sample code gives no response on the serial monitor. Is there anything I can do to determine what the problem is?

jpliew, 2015-08-25:
AWID SR-2400 Specification
"***Determined by AWID’s credentials programmed
with 26 bits to 50 bits***""

Please make sure that the unit is emitting 26 bits. This library only works with 4,8,26 and 34 bits wiegand.

Your schematic looks good, please also make sure that 5V is enough to power the reader. Most readers are at least 9V and above.

If nothing works, you might need a scope to check the pulses.

Cheers
JP

eoinverling, 2015-08-26:
Hi, 

the SR-2400 is powered and if I put in a "serial.println(_bitCount);" in the bool WIEGAND::DoWiegandConversion () function .... depending on what I set the baud rate to, I get different values (when I swipe a fob over the reader)

is there a timing / baud rate I should be using?

monkeyboard, 2015-08-26:
Hi,

I can see that you are powering the reader from Arduino's 5V. This 5V is supplied by a LDO that has only little current. That could be the reason why you have intermittent bit count.

How about powering the reader from external supply?

Also what baud rate are you talking about? There is no baud rate in weigand. 

Have you set your reader to to send the correct bit count that this library can detect?

Cheers
JP

jpliew, 2015-12-22:
No response from poster, closing.

closed 2015-12-22 by jpliew