monkeyboard/Wiegand-Protocol-Library-for-Arduino #10: "Wrong" card value detected - UHF Mid Range Reader

ecocomputerdesarrollo, 2015-10-23:
Hi.

We have the following setup:

* Arduino UNO.
* Self powered  [UHF Mid Range Reader](http://www.kimaldi.com/kimaldi_eng/products/rfid_systems/rfid_uhf_tags_and_readers/rfid_uhf_readers/rfid_uhf_readers_for_industrials_solutions/uhf_mid_range_integration_reader)

Pinout is made on ports 2 & 3 on the UNO.
Reader is set to output card value (EPC-GEN 2- Tag) through wiegand interface with wiegand26 protocol.

Testing on the reader program itself we get a 16byte tag as follows:
E2 00 41 25 24 0B 02 00  04 30 ea f9

And on arduino we only get (through the library wg.getCode() ) the mid 4byte value:
Decimal: 604701184
HEX: 24 0B 02 00

There's any change we can do to make it read (at least) the last 8 bytes?
Meaning "24 0B 02 00  04 30 ea f9" as unsigned long.

jpliew, 2015-10-23:
Hi,

12 byte tag is more than 26bit code. Do you have the specification of the reader?

Rgds
JP

jpliew, 2015-12-22:
No response from original poster. Closing this issue.

closed 2015-12-22 by jpliew