monkeyboard/Wiegand-Protocol-Library-for-Arduino #12: Poor handling of previously unknown bit lengths values.

digitalentropy, 2015-12-15:
This is a great library, but the fact that it only handles W26 and W34 is unfortunate.  There are many different bit lengths used by various vendors, and it would be highly useful to have a mode where it just saves the Wiegand data as-is, and does not try to decode it.  The failure mode for using unknown card types should be more graceful. :-)

jpliew, 2015-12-22:
Great suggestion, I will work on this and commit a new change soon.

digitalentropy, 2015-12-22:
Thanks for the reply!  I should also mention that the code appears to assume too much about cards that are 26 and 34 bits long, and tries to handle quite a bit of decoding automatically.  While this might be fine for your original test cases, it clobbers and corrupts other data formats.  Because there are so many different vendors and formats, I suggest defaulting to raw recording, and moving the decoding functions to another sub-function.  This would allow greater flexibility not only for users of various formats, but also would make it easier to add new decoding profiles.  Realistically the library should not try to automatically figure out what card format is present (with the exception of key presses, which are more standardized, although there are at least four or five different formats for that as well).  Instead, I think calling a particular format decoding function on demand might make more sense.

Your thoughts?

monkeyboard, 2015-12-22:
Maybe just add a getRAWDATA() function, that way, use can decide if they
need the RAW data or decoded function.

digitalentropy, 2015-12-22:
Perhaps, but moving raw to a different function seems to imply that most cards are handled gracefully by the default code.  In my view at the moment it does not, and I have a couple of hundred access cards and readers from different vendors that support that view.

If the aim is to create a Wiegand protocol library, than it should be a Wiegand library first, and a card format decoder second.

I hope that folks don’t misinterpret my suggestions as criticisms.  I really appreciate all of the work in the library so far, and the code is considerably cleaner than other implementations I’ve seen, but there has never been a single *good* Wiegand library.  All of the code I’ve seen has been hacked up just enough to get by for the specific purpose the author intended it for, and then left to become deprecated or broken.  This library is the best option available at the moment.

monkeyboard, 2015-12-22:
Hey mate,

Don't worry about the criticism, it is always criticism that makes
something better.

Before your message I was about to re-look and re-code the library, it is
getting a bit messy. The reason why "the code appears to assume too much
about cards that are 26 and 34 bits long," because it was originally
created for 2 of the readers I have, one is 26bit and the other one 34bit,
that's why I named it "Wiegand 26 and Wiegand 34 library for Arduino",
there were a lot of consideration that I did not put in.

Looks like it is a good time to re-code a new one with proper consideration:

I was suggesting the following

capture raw wiegand data, print if necessary
if needed user call decode() and function returns known data, if unknown,
function returns unknown data, showing the data and number of bits.

contributors can added different decoding functions in a different H file.
What do you think?

digitalentropy, 2015-12-22:
Yes, I think moving decoding to a separate header file may work well, especially if the format is well documented in the comments.  It would encourage others to add new formats as needed.

As far as the main function goes, perhaps the type of data returned can be passed into the function? Something along the lines of wg.getCode(RAW) or wg.getCode(H10301), etc.  There’s a useful thread that touches on the names of some of the data formats here: http://www.proxmark.org/forum/viewtopic.php?id=1653, but I may have some data sheets and documents that outline them more formally.  I just have to dig around.

I was also considering that the wg.available() function could could return the type of data available, e.g. CARD (for a credential) PIN (for a key press) or UNKNOWN if wiegand data is available in the buffer but it cannot be identified due to an unexpected bit length.

Not all of this needs to be written immediately of course, but if we’re going to re-structure the code anyway it provides a good opportunity to think forward a bit.

Just to give you a bit of a starting point, I happen to have an iCLASS tag that returns 142 bits of Wiegand data when presented to a reader.  I don’t know exactly why yet as it is quite a unique tag, but it’s precisely the kind of edge case that should be considered when writing failure modes in the code.

In my opinion if there is data in the buffer, it should be made available and not discarded.  There is rarely “noise" on the line unless there is something very wrong with the hardware design or with the wiring.

digitalentropy, 2016-01-23:
Any idea when you might have some cycles to throw at this?  I'm happy to contribute and test as I can but known for my coding skills I am not. :-)

jpliew, 2016-02-10:
Hi @digitalentropy could you help do a quick test before I work on the third party decoding please? Especially if you have readers that send out odd Wiegand.

Cheers
JP

jpliew, 2016-02-10:
https://github.com/jpliew/Wiegand-NG-Multi-Bit-Wiegand-Library-for-Arduino

digitalentropy, 2016-02-10:
Hey JP!

Great to see activity again!  I'll be back at the lab in a few days.  I've got tons of readers and credentials from different vendors to test with, and I'll let you know how it goes.

Regards,
Babak

Netoperz, 2017-07-24:
I have got this one:

https://www.genway.pl/product/attachment/08564f3fe4e658a1d8ac86dabd2cc900/pl_PL/vidi-ac-2cs-ang.pdf

Reading tags and cards works fine, but when it comes to the pin input sometimes it is not getting every keypress. when doing it slow it works in 90% of cases, but when fast nat always.

jpliew, 2017-08-02:
@Netoperz your reader has 3 modes for key press wiegand. 4, 8 or 10 bit. This library only support 4 bit and the 4 bit might not even match your reader's format, I am not sure because the PDF from your reader does not state what format it is.

Could you try with the Wiegand NG https://github.com/jpliew/Wiegand-NG-Multi-Bit-Wiegand-Library-for-Arduino  and let me know the output in binary please? For example

Press all the number and capture the binary output from Arduino console.

Ideally if you can capture the wave form of each keypress using a scope, that will help community update this library to support your reader.

Hope this helps.

jpliew, 2017-10-21:
If you have a device with bit length not covered by this library, please try the new multi-bit decoder https://github.com/jpliew/Wiegand-NG-Multi-Bit-Wiegand-Library-for-Arduino

closed 2017-10-21 by jpliew