monkeyboard/Wiegand-Protocol-Library-for-Arduino #14: Error compiling example

nvangs, 2016-02-19:
Hi,
Im getting this error when trying to compile your example:

```
\Arduino\libraries\Wiegand-Protocol-Library-for-Arduino-master\Wiegand.cpp: In member function 'void WIEGAND::begin()':
\Arduino\libraries\Wiegand-Protocol-Library-for-Arduino-master\Wiegand.cpp:37: error: 'NOT_AN_INTERRUPT' was not declared in this scope
```
Can you help me? Thanks.

monkeyboard, 2016-02-19:
Hi,

I don't have access to a computer now, try to rename your library name
following our readme. It should be arduino/libraries/wiegand

The current folder of yours has a very long name and I presume you
downloaded the zip file directly from github.

Hope this helps.

Cheers
JP

nvangs, 2016-02-19:
Hi JP,

Thanks for your reply. Renaming the library didn't help.

```
C:\Arduino\libraries\wiegand\Wiegand.cpp: In member function 'void WIEGAND::begin()':
C:\Arduino\libraries\wiegand\Wiegand.cpp:37: error: 'NOT_AN_INTERRUPT' was not declared in this scope
```

monkeyboard, 2016-02-19:
Check this out.

monkeyboard, 2016-02-19:
https://forum.arduino.cc/index.php?topic=353602.0

nvangs, 2016-02-20:
It did the trick! 

Thanks for a great library!

closed 2016-02-20 by nvangs