monkeyboard/Wiegand-Protocol-Library-for-Arduino #15: No Data Reported

exovaria, 2016-04-06:
Setup: 
- WiegandTest example sketch
- Arduino Uno
- Arduino Mega 2560
- HID ThinLine II Prox Reader

Behavior:
Sketch is loaded successfully to Uno.
HID ThinLine II reader connected to Arduino Uno using 5V from from Uno, Green (Data 0) connected to pin 2, White (Data 1) connected to pin 3, and black (GND) to ground. Upon powerup, the reader beeps and changes led state from red to green twice in a row, then a third time a half second apart. A single line, often "Wiegand HEX = 5, DECIMAL = 5, Type W4" is displayed over serial monitor (9600 bps) upon powerup. Subsequent card reads flash the led and beep but do not produce any output over serial. 

Attempts to Fix:
- Unplugged and replugged all connections
- Swapped Uno for Mega 2560
- Swapped Data 0 and Data 1
- Unplugged and restored power by removing 5V line (this causes the "Wiegand HEX = 5, DECIMAL = 5, Type W4" to appear often, but not always).

Any ideas about what could be going on? Please let me know. Thanks!

jpliew, 2016-04-06:
Hey @exovaria ,

1. Please make sure the HID Thinline II is the wiegand version.
2. Please make sure you are powering the reader from external power source, Arduino's power is limited.
3. If the Card produces wiegand other than 4, 26 or 34 bit, this library will not work. Then try out our new Wiegand NG that has unlimited bit. It is still beta though.
https://github.com/jpliew/Wiegand-NG-Multi-Bit-Wiegand-Library-for-Arduino

Hope this helps, and please report back your result.

Cheers
JP

exovaria, 2016-04-06:
Wow, that was fast!

1. Confirmed Wiegand version.
2. Switching to external power had no effect
3. This corrected it. Turns out these cards I'm using have 35 bits of data on them. That leads me to believe that the reads I got on power up were artifacts. Good call on the beta!

Thanks again for such a quick response.

closed 2016-04-06 by exovaria