monkeyboard/Wiegand-Protocol-Library-for-Arduino #17: UHF Reader Wiegand Doubled Value

mbal2002, 2017-05-17:
Hi

I got new UHF Long Range Reader and I can Read and Write on the Tags without any problems n the Card Number Block
The Numbers is was 1001 , 1002 , 1003 ...etc

The Problem is the Control Panel (ZK C3-400) get the Numbers  doubled  2002 , 2004 , 2006
the First Testing it was Ok at my home but after Fixing  it become like that

I dont Know does the (Pulse Width or Pulse Interval) Value will affect and what is the Right value

Please Help

jpliew, 2017-05-17:
Hi,

From your message, I am unable to understand what your problem was and what you intend to do.

The ZK C3-400 itself is a Wiegand reader, so how does this library work with the ZK C3-400 ?

If you can explain clearly and attached the diagram and code of what you wanted to do, then people might be able to help you. 

Cheers

mbal2002, 2017-05-18:
ZK c3-400  is Control Panel

get the Tags ID from long range UHF External Reader

The External Reader is Managed with SDK Software and used to Read and Write on the Tags
and also can manage the Wiegand Output Parameters

[image]
[PDUM33_E.pdf](https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/files/1010395/PDUM33_E.pdf)

The Reader Output is Wiegand34  connected to the C3-400 controller 
 the Problem is the Controller get the Reading mismatch the correct Data in the tag

My Question is  what may affect the Transfer Wiegand Data and make the Cards numbers  doubled the card number 1001  read by the controller 2002

The Distance between the Reader and the Controller only 3 meter and i Check the Connections and Protocols matching at both side Wiegand34

The Only Thing ididnt Change or try is the Pulse Width  and the Pulse interval  Parameter at the Reader Output because i dont know what they Affect exactly and what is the Proper Value should be

jpliew, 2017-05-18:
Ok, lets put something straight here, can we agree that the issue that you are currently facing has nothing to do with this Arduino Weigand Library on github?

1. I have not seen for Wiegand34, the card number is only 4 digit in your case 1001 is too short. Is this a HEX or DECIMAL? How do you know it is 1001? 

2. How about first, follow the instruction on PDUM33_E.pdf and use either serial or USB connect to the reader and scan the card, check the HEX of the card and see what is the card ID in HEX first.

3. Then also check D0 and D1 pin from the reader is connected properly to D0 and D1 or the controller. 

4. If still not working, you need to use an oscilloscope to capture the pulses of D0 and D1 and post it here, this is an example how it looks like http://www.monkeyboard.org/images/tutorials/wiegand/wiegand-25ms-break.gif

jpliew, 2017-06-20:
No response for 1 month, will be closing this issue as it looks like it is not an issue related to this library.

closed 2017-08-02 by jpliew