monkeyboard/Wiegand-Protocol-Library-for-Arduino #21: How can I get more characters in each reading?

deivid77, 2017-08-13:
I'm only getting 6 characters instead of 10.

Best regards
David Blanco

jpliew, 2017-08-13:
Hi @deivid77 it is hard to help without telling the community your reader (datasheet), your card/tag. This library only work with 8 bit, 26bit and 34 bit wiegand.

Ideally it will be easier to diagnose your problem if you can capture the waveform like this
[image]

If you don't have a scope to capture the waveform, please try this new library and report back the result
https://github.com/jpliew/Wiegand-NG-Multi-Bit-Wiegand-Library-for-Arduino

This new library can do unlimited bit length.

deivid77, 2017-08-14:
Hi JP,
I forgot to tell you the tech specs, my mistake!

I'm using a GP60A RFID Reader:
[](http://www.promageurope.com/products/rfid-readers-and-writers/gp60-long-range-rfid-reader-gp60a.htm)

This is a RS232 and Wiegand 26 reader, among other things.
With RS232/TTL I'm getting 10 ASCII chars, but with Wiegand I'm only getting the last 6 chars.
I'm starting to see that maybe it's right: 4 bits x 6 chars = 24 + (STX) + (ETX) = Wiegand 26 bits

[GP60A EN Rev.E.pdf](https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/files/1221427/GP60A.EN.Rev.E.pdf)

With MonkeyBoard Library I'm getting:
 Wiegand HEX = 204306, DECIMAL = 2114310, BINARY = 1000000100001100000110, Type W26

With NG-Multi-Bit Library I get:
RAW Binary=00010000001000011000001101

I have a DSO 138 Scope but I'm afraid I'm too much newbie to get your pulses and freeze them.

jpliew, 2017-08-14:
Hi David,

Thanks for the reply.

You didn't tell me what was the data you got for the 10 ASCII chars so I can't compare. But from the datasheet, looks like the results you got from this library and NG library are correct and matching the reader's datasheet page 3,

Wiegand interface (26-bit format)

However how the data is being translated to serial interface data (10x "hex" ASCII chars), I am not sure, but if you can also post the 10x ASCII result, we can at least guess how the translation works.

Cheers
JP

deivid77, 2017-08-14:
Hi again JP.

Here you are the ASCII chars:
(STX) 4C00204306 (CR) (LF) (ETX)

I hope it helps.

Thank you so much!
David Blanco

jpliew, 2017-10-16:
@deivid77  I thought I have responded to you but looks like it did not show up here.

Basically you have the same data there.

From RS232 ASCII
(STX) 4C00204306 (CR) (LF) (ETX)

From Wiegand 
204306

Because your reader only output 26bit, minus two bits, the data is 24 bit
There were 3 bytes on 204306, total of 24bits. 

Looks correct to me.

deivid77, 2017-10-23:
That's what I thought, thank you @jpliew

closed 2017-10-23 by deivid77