monkeyboard/Wiegand-Protocol-Library-for-Arduino #24: Missing one bit in serial number

Thiagodcfarias, 2017-12-05:
Hi,
 i can get the serial number of my RFID using the example in this library. However, i´m missing the first bit of serial number and i don´t know how to fix it.

My RFID uses 34 bits, 32 for serial number and 2 for parity, but when i code "Serial.print(wg.getCode(),BIN);" i recieve just 31 bits :(

jpliew, 2017-12-05:
@Thiagodcfarias could you run the example code and paste the full result here please?

Thiagodcfarias, 2017-12-11:
Hi,
the code:

#include <Wiegand.h>

WIEGAND wg;

void setup() {
	Serial.begin(9600);  
	wg.begin();
}

void loop() {
	if(wg.available())
	{
		Serial.print("Wiegand HEX = ");
		Serial.print(wg.getCode(),HEX);
		Serial.print(", DECIMAL = ");
		Serial.print(wg.getCode(),BIN);
    Serial.print(", BINARIO = ");
    Serial.print(wg.getCode());
		Serial.print(", Type W");
		Serial.println(wg.getWiegandType());    
	}
}

the result:
Wiegand HEX = 51D85DFA, DECIMAL = 1010001110110000101110111111010, BINARIO = 1373134330, Type W32

Thiagodcfarias, 2017-12-11:
So, if the RFID´s serial number starts with 0, the translation to 10´base is equal to the number of my RFID card, in the other hand, the numbers are different when starts with 1.
Because of that, i think that we just can see the numbers through a window

jpliew, 2017-12-11:
@Thiagodcfarias there is nothing wrong with your result. The HEX, BIN and DECIMAL results are identical.

`Serial.print()` binary will not print any leading 0.

closed 2018-01-02 by jpliew