monkeyboard/Wiegand-Protocol-Library-for-Arduino #26: Cant seem to get the card read

akeilox, 2018-01-27:
I have a generic EM Wiegand 26/34 bit reader.
The cable pin outs are standard, 12v, Gnd, D0, D1, purple (when connected to gnd turns it to 34 bit) and buzzer, as shown in picture https://imgur.com/a/BvFgg

I have setup the sketch and uploaded to Arduino Uno, code runs and awaits  wg.available()
I have the 12v and GND connected to 12v +/-, and 
only D0,D1 connected to Arduino uno io pins 2 and 3 (i have not connected GND to arduino gnd pin, as its already connected to 12v -)

when i tap an EM card, the reader beeps as card detected, but there is no output on the serial monitor.
I can see the reader not responding to non EM cards (like Mifare) and always beeps when EM card tapped.

I cant seem to find why the example sketch wq.available() doesnt work in this case.
The reader is generic, arduino uno works, D0 (green) and D1 (white) connected to pins correctly, 2 and 3.

I am not sure what could be the issue. Is there any pointers that You can share or alternative troubleshooting advise?

jpliew, 2018-01-27:
Hi @akeilox could you try connect the ground from the reader to Arduino please?

If still not working, please try https://github.com/jpliew/Wiegand-NG-Multi-Bit-Wiegand-Library-for-Arduino and post your result in binary here.

akeilox, 2018-01-28:
Hi @jpliew 
Thank you for your patience. When I put another wire from 12V - (gnd) to arduino Gnd , it worked!

Does it mean the GND connection is a must for all wiegand readers? Like even i use esp8266 etc

akeilox, 2018-01-28:
Thanks @jpliew 
I'm new to Wiegand readers, and had a wrong understanding/assumption of D0 and D1 lines.
GND connection is a must, whether its Arduino or ESP board, that I understand.

Thank you for taking time clarify, appreciated!

closed 2018-01-28 by akeilox