monkeyboard/Wiegand-Protocol-Library-for-Arduino #28: How to define the D0 and D1 pins when calling the init for wiegand in sketch without changing the code in library by hand ?

Netoperz, 2018-03-06:
In the library in Wiegand.cpp you can set the proper pins for the Wiegand interface.
Is there a way to define those pins in sketch ? if I have two different designs i must check what is set in Wiegand.cpp, also it makes problems when sharing the code over git, because if someone is using default configuration the code actually will not work and that gives another "complication" when working with many people with the same code.

`void WIEGAND::begin()
{
#ifdef digitalPinToInterrupt
  // newer versions of Arduino provide pin to interrupt mapping
  begin(2,digitalPinToInterrupt(2),3,digitalPinToInterrupt(3));
#else
  begin(2,0,3,1);
#endif`

So as i'm not a specialist in terms of programming i'm asking how can i init wiegand interface in sketch and point it to the proper pins if they are different than the ones hardcoded in library?

can i define the pins when calling the init for wiegand ?
wg.begin();

and how exactly, and if not, than will someone add that functionality for this library ? It would be very very useful.

Thanx

jpliew, 2018-03-06:
Hi @Netoperz yes, setting the pins in sketch is supported, please read the README.md but please take note that not every pin works, only those that has interrupt supported can work.

README.md is here
https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino

Hope this helps.

jpliew, 2018-03-07:
@Netoperz maybe easier to see from the code, there is another declaration in [`Wiegand.cpp`](../blob/master/Wiegand.cpp) just right below the code you showed in your question

```c
void WIEGAND::begin(int pinD0, int pinD1)
{
  begin(pinD0, pinD0, pinD1, pinD1);
}
```

This init allows user to use their own pin in sketch.

Netoperz, 2018-03-11:
works fine, thank you :)

closed 2018-03-11 by Netoperz