monkeyboard/Wiegand-Protocol-Library-for-Arduino #32: 'hang' after receiving number of bits different from 8, 26 and 34

bilbobg, 2018-09-04:
If for some reason the library receive bit counts different from 8, 26 and 34 the _bitCount variables never resets and  DoWiegandConversion function does not execute anymore. Simple fix is to put _bitCountA=0; _bitCountB=0; _bitCountC=0; in the else statement for "if ((_bitCountA==26) || (_bitCountA==34) || (_bitCountA==8) || (_bitCount...".

jpliew, 2018-09-04:
Hey @bilbobg ,

The code took care of anything over 25ms, and _bitCount actually does reset starting here

https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/blob/1c2b580293768f01abf932c7830ea8ec0ebf0a7d/Wiegand.cpp#L180

bilbobg, 2018-09-05:
ok, probably is my fault :) the version that I have has only "return false;"

closed 2018-09-05 by bilbobg