monkeyboard/Wiegand-Protocol-Library-for-Arduino #35: How can I set the library to work only with 8 bit wiegand, and 26 bit wiegand

Netoperz, 2019-03-06:
Is there a way to set the library to work with only one or two types ow wiegand encoding ? for example to ignore the 4 bit , and wait only for 8 bit or 26 bit ?

jpliew, 2019-03-06:
@Netoperz if you wiegand does not send out 4 bit data, the library will not report 4 bit data.

If you wiegand is sending out 4 bit, 8 bit and 26 bit at the same time, you can use 

getWiegandType() to check and drop the data if it returns a 4

example

```
if ( (wg.getWiegandType() == 8) || (wg.getWiegandType()==26) 
{
 // do your calculation here
}
```

anything other than 8 or 26 will be ignored.

Netoperz, 2019-03-06:
@jpliew 
That's excellent, i'll try that tomorrow, also i'm getting oscilloscope tomorrow,  so i'll be able  to test the signaling regarding those extra bits that i'm getting on keypads.
Last resort is to use OSDP keypads, but i cannot find any info about arduino lib. that works with OSDP v1/2  if you have any knowledge about that please throw me a bone :) thanx

closed 2019-09-08 by jpliew