monkeyboard/Wiegand-Protocol-Library-for-Arduino #36: Different readers return different value

iBobik, 2019-05-27:
I tried to read tags with different readers, but it seems like decoding is reader-specific.

This two values are the same tag, but different reader:
7384CF
8C7B30

For both readers it prints W26 reader type.

Is there some way how to recalculate it, so I can use different readers in one access system?

jpliew, 2019-05-27:
Hi @iBobik ,
Please provide a screen capture from an oscilloscope for D0 and D1 for both readers. The screen should look like this.

[image]

From the image, I can then see how many 0 and how many 1 and compare with the two readers.

Some readers have different specifications, even though the send data in wiegand, but the way the data is organise might be different.

Hope this helps.

jpliew, 2019-07-19:
@iBobik do you still need help? If not I will close this issue.

iBobik, 2019-07-19:
I postponed it on my side, but thank you for help. I will use it but do not know when.

JohnMcLear, 2019-08-25:
I can replicate this bug.

1. Plug in a reader, scan Mifare enabled card that has secure element (IE Mifare enabled P5/6//SLE77/78)
1. Get ID
1. Unplug Wiegand reader ~~~~ Don't restart usage logic or reload module / library etc.
1. Plug in different Wiegand reader
1. Scan same card.  You will notice a different ID.

First scan will be Mifare ID (4 bytes)

Second scan will be non-Mifare ID (one assumes ~NTAG ID) which is longer in length (Documented as 7 but displays as 5).

To add to this, this library **completely fails to scan UID or a non-Mifare enabled tag**.  While this is probably a separate issue it is something that needs documenting clearly in the README to ensure users who have non Mifare enabled Secure Element based cards (IE P5,6,77,78 -- which make up a huge amount of cards in circulation) are supported.  

The correct patch fix here is to query the SEID through an applet for a usable UID.   The correct long term fix is to bring in support for additional cryptographicly signed methods for getting a UID IE something that returns UID (from SEID) after 3Des handshake.

So to summarize:
  This library fails to consistently gather the correct UID of any Mifare enabled tag should the wiegand connected device be changed.
  This library fails to gather the correct UID of some cards in circulation at current (namely smart cards without Mifare support)

Thanks to the author for his work so far and ongoing, I hope we can prioritise these fixes because IMHO they are quite critical to a lot of use cases currently consuming this library.

jpliew, 2019-08-25:
Hey @JohnMcLear,

Thanks for your comment.

I am actually a bit confused here.

This library only reads wiegand data, it is not aware of what card being scanned. The card reader reads the card, process the ID and then send the value out as wiegand pulses.

It is the responsibility of the card reader to negotiate the RF part of the protocol with the card being scanned.

Hope this clears up the confusion.

Cheers
JP

JohnMcLear, 2019-08-26:
I commented here but including copy/paste here.

https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/issues/40

@jpliew Oh. So the wiegand device is the limitation, very interesting and this would explain why certain readers read different values from the same card. Wow, what a fucked up situation to be in!

So TLDR; if you use Wiegand, you can't be consistent w/ UIDs if you use different makes/models of readers because the Wiegand logic implemented changes between devices (Despite being a standardized protocol [citation needed]).

This has to be flagged as "wont fix" because you can't fix.  Basically the only fix here is to ensure that the same exact make/model/software version of the Wiegand device is used upon replacing a reader, anything else might return a different value :(

The crux of the problem is due to Wiegand protocol being ?improperly? implemented.

iBobik, 2019-08-26:
Is it theoritecally possible to investigate reader’s encoding algorithm from returned values of known card? Then we can implement converting of UIDs, so we can use different readers in one system without maintaining multiple UIDs for one user for each type of reader.

JohnMcLear, 2019-08-27:
> Before we go nuts over the this, it is better to checkout the specification of the card readers. As far I as know, there are card readers that will be able to be reconfigured for different types of data.
>
> In this industry, the card readers (mostly high end type) has to be reprogrammed to suit the different systems use to ensure consistency.
>
> -

Yeah I imagine that's the problem.  I'm using super sketchy / budget readers.

jpliew, 2019-08-27:
Before we go nuts over the this, it is better to checkout the specification of the card readers. As far I as know, there are card readers that will be able to be reconfigured for different types of data.

In this industry, the card readers (mostly high end type) has to be reprogrammed to suit the different systems use to ensure consistency.

jpliew, 2019-09-08:
@iBobik I double check your data again and I can confirm that when you switched over the reader, the D0 and D1 are swapped.

```
7384CF
8C7B30
```
these values are NOT or each other, see binary data below

```
011100111000010011001111
100011000111101100110000
```

So your two readers producing the same data.

iBobik, 2019-09-08:
Wow, I feel so stupid now. :-) Thank you a lot

Jan Pobořil
https://honza.poboril.cz/