monkeyboard/Wiegand-Protocol-Library-for-Arduino #37: Reversed Byte Order

giotec, 2019-06-27:
The actual card value is 12 34 56 78, however the library spits out 78 56 34 12. Is there a way to reverse the byte order within the library?

jpliew, 2019-06-27:
@giotec the library does not do any translation, it just print out exactly what was captured on D1 and D0. There some reader that will reverse the data of the card. You might want to check the setting of the reader if it is BIG or LITTLE ENDIAN.

If you can't configure your reader, you can check out how to convert between the endians here
https://stackoverflow.com/questions/2182002/convert-big-endian-to-little-endian-in-c-without-using-provided-func

Hope this helps.

giotec, 2019-06-28:
Cheers,

The second answer in that link worked perfectly.

added #include <byteswap.h> and put __bswap_32 (wg.getCode())

closed 2019-06-28 by giotec