monkeyboard/Wiegand-Protocol-Library-for-Arduino #39: is possible LED control?

joztoz, 2019-07-18:
Hello, my reader have also LED, but I have only one wire, please how I can control color of led?
Second wire is marked as a Base, but I dont understand if its tamper or buzzer, I think, there wire marked as LED is for green/red led and buzzer.
Please how can i control it? connected to GND or VCC is not doing anything.
Thank you

jpliew, 2019-07-18:
Hi @joztoz please take a photo of your reader and provide datasheet/specification of the reader. Thanks

joztoz, 2019-07-19:
Hello, It is this card and keypad reader:
https://dahuawiki.com/Access_Control/Products/ASR1101M

problem is that its only this one specification on internet.
There is one wire (brown) for Led control and for buzzer I think.

jpliew, 2019-07-19:
@joztoz attached is the user manual

[dhi-asr1101m-d__i_en.pdf](https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/files/3409938/dhi-asr1101m-d__i_en.pdf)

Looks like you can only control the LED, brown wire.

This reader need to be power up from 12VDC. 

Try connect the brown wire to 12VDC. If still not working you might want to get support from your supplier.