monkeyboard/Wiegand-Protocol-Library-for-Arduino #40: UIDs not available for GlobalPlatform based Secure Elements that don't include Mifare

JohnMcLear, 2019-08-25:
https://github.com/esprfid/esp-rfid/issues/292

Includes proposed fix.

jpliew, 2019-08-26:
Hey @JohnMcLear I think there is a misunderstanding somewhere. 

This library reads wiegand data, it has no control over how to read a card, how to read different types of card.

The card reader that has wiegand output reads the card, interpret the data of the card, and send out the data it deemed correct to wiegand output, this library reads the wiegand output and display the content of the wiegand output.

Cheers
JP

JohnMcLear, 2019-08-26:
@jpliew Oh.  So the wiegand device is the limitation, very interesting and this would explain why certain readers read different values from the same card.  Wow, what a fucked up situation to be in!

So TLDR; if you use Wiegand, you can't be consistent w/ UIDs if you use different makes/models of readers because the Wiegand logic implemented changes between devices (Despite being a standardized protocol [citation needed])

closed 2019-08-26 by JohnMcLear

jpliew, 2019-08-27:
@JohnMcLear, Wiegand is a standard protocol but how the content is encoded in this standard is depending on the device.

Just like TCPIP is a standard, but how application send the content (http,smtp,ping, etc) is depending on the application.