monkeyboard/Wiegand-Protocol-Library-for-Arduino #43: What types of pins do I need to connect D0 and D1 to an ESP32 ?

GeorgeFlorian, 2019-12-02:
Hello !

Can I randomly choose two GPIO from the ESP32 pins or do they have to be specific pins, like UART pins ?

For example, if I want to use the Serial interface on the ESP32-EVB I will use GPIO 4 and GPI 36 specifically:
```
#define RX1 36
#define TX1 4
```

Do D0 and D1 have the same requirements ? Or I can choose whatever pins I have available ? Do they have to be input or output or both ?

Thank you.

jpliew, 2019-12-02:
from ESP32's documentations, and GPIO will do.

GeorgeFlorian, 2019-12-03:
> from ESP32's documentations, and GPIO will do.

I am sorry, I didn't understand a thing from what you've written.

jpliew, 2019-12-03:
Yes, any GPIO for ESP32 will be able to work. When you use the library, the initialisation will make the GPIO pin to be INPUT and with INTERRUPT.

mrarmyant, 2020-02-24:
just tried the library with an esp32, seems to read a ton of items with no cards in front of it.  Does it need pullup resistors?

jpliew, 2020-02-24:
> 
> 
> just tried the library with an esp32, seems to read a ton of items with no cards in front of it. Does it need pullup resistors?

Check with an oscilloscope first the D0 and D1 signal from the reader, also for ESP32 you need a voltage level converter to convert the voltage of D0 and D1 to the input level of ESP32.

Hope this helps.

mrarmyant, 2020-02-24:
Awesome thanks.  I bet the voltage is the issue I thought the mfr said it was 3v but who knows.  I wonder of any are out there that work without a converter -.-

closed 2020-02-25 by GeorgeFlorian