monkeyboard/Wiegand-Protocol-Library-for-Arduino #46: Feature Request - Pin Change Interrupt Support

zacaustin, 2020-02-11:
I was wondering if it would be possible to add Pin Change Interrupt support to this library to enable a Wiegand reader to be connected to another pair of pins on the board?

Currently working on a project where I need to use the Pins 2 and 3 on an Arduino Uno for other purposes but can't use a Mega to get additional Hardware Interrupt Pins.

I would have a go myself but I'm not quite at the skill level to implement this.

fjld93, 2020-09-15:
Hi, the quick solution that I used was use the PinChangeInterrupt library by Nico Hood, and then, change the "private" to "protected" in the header file, in this way, we can use the variables in a derivated class.

After that, i made a derivated class and just overload the begin method, just like:

`#include <Wiegand.h>

#include <PinChangeInterrupt.h>
#include <PinChangeInterruptBoards.h>
#include <PinChangeInterruptPins.h>
#include <PinChangeInterruptSettings.h>

class WiegandPCINT : public WIEGAND {

  public:
  void begin(int pinD0, int pinD1){
    _lastWiegand = 0;
    _cardTempHigh = 0;
    _cardTemp = 0;
    _code = 0;
    _wiegandType = 0;
    _bitCount = 0;  
    pinMode(pinD0, INPUT);          // Set D0 pin as input
    pinMode(pinD1, INPUT);          // Set D1 pin as input

    attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(pinD0), ReadD0, FALLING);
    attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(pinD1), ReadD1, FALLING);
  }
  
};`

With this changes, you can use any pin that allow PCINT. Note that PinD0 is the pin that will receive the "0" bits, and PinD1 will receive the "1" bits.

Regards.

jpliew, 2021-02-15:
Please take a look at this

https://github.com/jpliew/Multi-Reader-Wiegand-Protocol-Library-for-Arduino