monkeyboard/Wiegand-Protocol-Library-for-Arduino #48: randomly reporting card with "0" code and differnt card lengths

gshiftlabs-dburke, 2020-05-14:
I started using the library and it's reporting random card numbers and lengths every once in a while. (randomly)

it usually reports "0" for a card number and various lenghts, but does occasionally give some strange card numbers "4"  "5" "ffffff" etc.  

all the wires are soldered in with the exception of the cable from the reader.. 

I'm using an esp32 nodemcu 

has anyone seen this  before?

jpliew, 2020-05-14:
Hi @gshiftlabs-dburke 

Looks like noise on the lines or power.

Please check that

* there is a 5V to 3.3V level converter from the reader's D0 and D1 to ESP32
* proper voltage to supply to the reader, example 12V for most of the readers
* the connections from the reader to the 5V to 3.3V level converter are properly secured. Not using breadboard or twisting the wires

Hope this helps.

gshiftlabs-dburke, 2020-05-15:
Level converter is being used
Voltage to reader is within spec. Using 5V (5V  to 16 V supported) 
all connection are hard-wired and soldered, no dupont cabled used
a good quality 5V usb power supply is being used.

jpliew, 2020-05-15:
You might need an oscilloscope to view the D0 and D1 wave form.

gshiftlabs-dburke, 2020-05-15:
I'll try upping the voltage on the card reader or swapping it out (I have a few I can try)

Thanks!

yesiamthatman, 2020-07-19:
I had been following this issue since I had the exact same issue.  I'd been using 5V on a HID iClass SE R10 scanner and had been experiencing the "0" and other anomalous scans.  I'd tried two different level shifters and switching to a voltage divider circuit, but nothing could quite shake it out.  I've since switched to 12V for the scanner, went back to a level shifter, and I was able to get 50 straight scans without issues.  So far. Obviously, I'm going to keep testing this thing out, but so far 12V seems to have been rock solid.

closed 2024-09-28 by jpliew