monkeyboard/Wiegand-Protocol-Library-for-Arduino #50: Same card same reader different data

AuspeXeu, 2020-09-24:
I am using your library with [this](https://www.amazon.de/gp/product/B07DK9R7K4/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&th=1) card reader. However for the same card it gives me a different output every single time and the type is detected as `W34` for some reason. What am I doing wrong?

```
11:11:07.914 -> Wiegand HEX = 854B50F5, DECIMAL = 2236305653, Type W34
11:11:14.045 -> Wiegand HEX = A1EA42BB, DECIMAL = 2716484283, Type W34
11:11:19.744 -> Wiegand HEX = 41D2953B, DECIMAL = 1104319803, Type W34
11:11:23.110 -> Wiegand HEX = 854B50F5, DECIMAL = 2236305653, Type W34
11:11:28.462 -> Wiegand HEX = A9CA82AB, DECIMAL = 2848621227, Type W34
11:11:31.486 -> Wiegand HEX = 8ACA81AB, DECIMAL = 2328527275, Type W34
11:11:34.393 -> Wiegand HEX = 43455475, DECIMAL = 1128617077, Type W34
```

jpliew, 2020-09-24:
Please check if you are 

1. Providing 12V DC to the card reader
2. GND of the 12V power supply is connected to the card reader and also to the Arduino
3. D0 and D1 are connected properly.
4. What Arduino board are you using? If you are using a 3.3V Arduino, you need a 3.3V to 5V logic level converter

From your data, it looks like noise.

AuspeXeu, 2020-09-24:
1. yes, via an external power supply
2. I am not sure how this is to be done since I am using an ESP8266 board (like a wemos)
3. I would think so
4. [This](https://www.amazon.de/gp/product/B0754N794H/ref=ppx_yo_dt_b_asin_title_o02_s02?ie=UTF8&psc=1) board

jpliew, 2020-09-24:
You need a 3.3V to 5V level converter.

ESP8266 IO PIN ----->3.3V------>5V---->D0 of reader
ESP8266 IO PIN ----->3.3V------>5V---->D1 of reader
ESP8266 GND----->GND of the level conveter--->GND of the READER----> GND of the power supply

AuspeXeu, 2020-09-24:
Just for my understanding, is this because inputting 5V into the GPIO will over saturate it?

jpliew, 2020-09-24:
Hard to say, it best to get everything right first.  ESP8266 is 3.3V component.

From the data, I still think is noise causing it. Ideally you have an oscilloscope that can view the waveform of D0 and D1. This will give you a clearer picture what is causing the issue.

The other thing is the card reader could be faulty, but without a scope, again, hard to say.

AuspeXeu, 2020-09-24:
I see. Unfortunately I don't have access to a scope. Can you recommend an alternative to the `ESP8266` that might work on 5V?

jpliew, 2020-09-24:
If you have an Arduino UNO or Leonardo, you can give it a test, they are all 5V. By doing this you can confirm if the reader is good or bad.

AuspeXeu, 2020-09-24:
One thing that I realised I forgot to mention is that the logs coincide with me holding a chip on the reader. Does this change your diagnostics of the problem?

jpliew, 2020-09-24:
Holding what chip?

AuspeXeu, 2020-09-24:
The rfid chip, sorry

jpliew, 2020-09-24:
Maybe you can explain clearer please. What did you actually scanned? Card or RFID chip?

AuspeXeu, 2020-09-25:
The output I get in my very first comment coincides with me holding an RFID chip at the reader.

jpliew, 2020-09-26:
Hey mate, I am lost.

Could you please answer my previous question, What did you actually scanned? Card or RFID chip? 

It is hard to debug your issue unless you can clearly explain in a way that one can "see" what you were doing. 

More questions:

If you were scanning RFID chip, did you make it yourself? 

If you were scanning RFID chip, have you try scanning a card?

AuspeXeu, 2020-09-28:
Sorry for the confusion. I have the following setup right now.

Parts:
[RFID reader](https://www.amazon.de/gp/product/B07DK9R7K4/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&th=1)
[ESP8622](https://www.amazon.de/gp/product/B0754N794H/ref=ppx_yo_dt_b_asin_title_o02_s02?ie=UTF8&psc=1)
[TXS0108E](https://www.amazon.de/gp/product/B07N7FFY2Q/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)
[Chips](https://www.amazon.de/RFID-Transponder-Chipschl%C3%BCssel-125KHz-Blau/dp/B005DR0AOG/ref=pd_bxgy_img_3/258-6156725-3720629?_encoding=UTF8&pd_rd_i=B005DR0AOG&pd_rd_r=ee7663d4-e8ed-4343-8e0c-4fb41a2d3b85&pd_rd_w=W1JcH&pd_rd_wg=rSyxx&pf_rd_p=4490d326-3239-499b-8c8a-0dd8d6ad582a&pf_rd_r=Z1BPJWBJ3RP1P4PP2E99&psc=1&refRID=Z1BPJWBJ3RP1P4PP2E99)

The RFID reader is connected to a 12V power source. The `ESP8622` is connected via USB to my laptop. `D0` and `D1` of the RFID reader are connected to the `B3` and `B4` of the `TXS0108E`. The `ESP8622`'s `D1`, `D2`, and `GND` are connected to the `TXS0108E`'s `A3`, `A4`, and `GND`, respectively.

The `ESP8622` is running the following code.
```
#include <Wiegand.h>

WIEGAND wg;

const int pinD0 = 5;
const int pinD1 = 4;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  wg.begin(pinD0, pinD1);
}

void loop() {
  // put your main code here, to run repeatedly:
  if(wg.available()) {
    Serial.print("Wiegand HEX = ");
    Serial.print(wg.getCode(),HEX);
    Serial.print(", DECIMAL = ");
    Serial.print(wg.getCode());
    Serial.print(", Type W");
    Serial.println(wg.getWiegandType());    
  }
}
```

However I don't get any output via the serial monitor.

jpliew, 2020-09-28:
1) Did you supply power to VA and VB of TXS0108E ?
VB is the READER side, should be 5V.
VA is the ESP side, should be 3.3V

2) When you order the RFID, which version did you pay for ? KR601 IC or KR601 ID ?

AuspeXeu, 2020-09-28:
1. I have no power supply connected to the TXS0108E the reader consumes 12V and I don't have a steady 5V stream available.
2. KR601 IC (whats the difference?)

jpliew, 2020-09-28:
1) TXS0108E need power to operate.  Without power it will not work
VB is the READER side, should  connect to 5V
VA is the ESP side, should connect to 3.3V

2) 
KR601 IC is W34 with 13.56Mhz 
KR601 ID is W26 with 125Khz

AuspeXeu, 2020-10-19:
I ordered an Arduino UNO now and connected as you specified.

D0 of reader > D2 of arduino
D1 of reader > D3 of arduino
external power supply VCC and ground > reader

And I flashed this program on the arduino
```
#include <Wiegand.h>

WIEGAND wg;

const int pinD0 = 2;
const int pinD1 = 3;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  wg.begin(pinD0, pinD1);
}

void loop() {
  // put your main code here, to run repeatedly:
  if(wg.available()) {
    Serial.print("Wiegand HEX = ");
    Serial.print(wg.getCode(),HEX);
    Serial.print(", DECIMAL = ");
    Serial.print(wg.getCode());
    Serial.print(", Type W");
    Serial.println(wg.getWiegandType());    
  }
}
```

However I don't get anything printed on the serial monitor....

jpliew, 2020-10-19:
Hi, can you please take a photo of your setup? 

From your description, you are missing GND of the READER to GND of the Arduino. Please see image

![https://camo.githubusercontent.com/6079648d2a60e606532500a0c98decb326b362ab/687474703a2f2f7777772e6d6f6e6b6579626f6172642e6f72672f696d616765732f7475746f7269616c732f77696567616e642f77696567616e645f61726475696e6f2e706e67](https://camo.githubusercontent.com/6079648d2a60e606532500a0c98decb326b362ab/687474703a2f2f7777772e6d6f6e6b6579626f6172642e6f72672f696d616765732f7475746f7269616c732f77696567616e642f77696567616e645f61726475696e6f2e706e67)

AuspeXeu, 2020-10-20:
This  solved my issue. Much appreciate you taking the time to help me. Even the ESP8266 works now.

closed 2020-10-20 by AuspeXeu

Wuckel01, 2022-04-10:
Hello,
 I have a strange effect.
I also have a KR601 IC is W34 with 13.56Mhz reader I connected it with a signal converter and the UID is reversed.
If I use a certified USB card reader, the UID is 1A:1A:16:1A
I neme KR601IC reader comes out HEX = 1A161A1A there is a solution

jpliew, 2022-04-10:
Hi @Wuckel01 this issue has closed. If you need to file and issue, you should create a new one.

From your description, unfortunately your reader is providing the bytes in a reversed order. This is not a problem, it is just the way the reader present the data. This is not the problem of this library either.

All you need to do is to swap the bytes from the result following suggestions from here
https://stackoverflow.com/questions/2182002/convert-big-endian-to-little-endian-in-c-without-using-provided-func

or alternatively, if you don't wish to do the above, you can use the USB reader read the UID and then reverse it manually and save the reversed UID into your Arduino sketch. Therefore when you compare you will get the same result.

Wuckel01, 2022-04-16:
Hello
Many thanks for the super fast reply that really helped.
if someone has the same problem, what works great for me.
`#include <Wiegand.h>

WIEGAND wg;

const int pinD0 = 3;
const int pinD1 = 2;
long unsigned num = "";
long unsigned swapped = "";
int i = 0;

void setup() {
  Serial.begin(9600);
  wg.begin(pinD0, pinD1);
}

void loop() {
   
  if(wg.available())
  {

    num = wg.getCode();
    swapped = ((num>>24)&0xff) | // move byte 3 to byte 0
                    ((num<<8)&0xff0000) | // move byte 1 to byte 2
                    ((num>>8)&0xff00) | // move byte 2 to byte 1
                    ((num<<24)&0xff000000); // byte 0 to byte 3
                    
    //Serial.print("Wiegand HEX = ");
    //Serial.println();
    //Serial.println(wg.getCode(),HEX);
     Serial.println(swapped,HEX);
    //Serial.print(", DECIMAL = ");
    //Serial.print(wg.getCode());
    //Serial.print(", Type W");
    //Serial.println(wg.getWiegandType());  
     
    }
  
}`