monkeyboard/Wiegand-Protocol-Library-for-Arduino #51: Connect two readers to one controller

AuspeXeu, 2020-11-18:
Was just wondering whether it would be possible to monitor two readers with the same controller. I tried the following code but it only logs reads from `Wiegand1`.

```
#include <Wiegand.h>

WIEGAND wg1;
WIEGAND wg2;

const int pinD01 = 2;
const int pinD11 = 3;
const int pinD02 = 10;
const int pinD12 = 11;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  wg1.begin(pinD01, pinD11);
  wg2.begin(pinD02, pinD12);
}

void loop() {
  // put your main code here, to run repeatedly:
  if(wg1.available()) {
    Serial.print("Wiegand1 HEX = ");
    Serial.print(wg1.getCode(),HEX);
    Serial.print(", DECIMAL = ");
    Serial.print(wg1.getCode());
    Serial.print(", Type W");
    Serial.println(wg1.getWiegandType());    
  }
  if(wg2.available()) {
    Serial.print("Wiegand2 HEX = ");
    Serial.print(wg2.getCode(),HEX);
    Serial.print(", DECIMAL = ");
    Serial.print(wg2.getCode());
    Serial.print(", Type W");
    Serial.println(wg2.getWiegandType());    
  }
}
```

jpliew, 2020-11-18:
This will not work as the library was not written to cater for multiple wiegand. I will put this in a feature request and work on it when free.

AuspeXeu, 2020-11-19:
C++ is not my daily driver so I am not super aware of how it works but what changes would be necessary for this to work? It's not like in OOP and another `Wiegand` object is instantiated and that's it?

jpliew, 2020-11-19:
There are two things that needed to be done.

1) Move all the variables into proper C++ class

2) When I first wrote this, it was meant for UNO, and UNO has only two interrupts. In order to support multiple interrupts, we need to use pin change interrupts.

AuspeXeu, 2020-11-19:
I just looked into the code it appears to me that it's already a C++ class, no?

AuspeXeu, 2020-11-20:
I can confirm my code from the first post in this issue works with two readers connected to an Arduino Mega.

jpliew, 2020-11-21:
Because the variables were designed for only 1 reader in the global, it will not completely work for 2 readers. In your case it worked because you swiped the cards one by one. 

I am going to leave this issue open so that others can see and also I try to allocate time to work on this.

jpliew, 2020-11-21:
The other thing, I am not sure how it worked on your test when you were using pin10 and pin11 for the second reader. Pin10 & 11 are not interrupt supported.

AuspeXeu, 2020-11-23:
Fixed the code. I actually used different pins.

jpliew, 2020-11-23:
Try scan two reader at the same time, you will get wrong value.

AuspeXeu, 2020-11-23:
I can confirm that yes.

AuspeXeu, 2021-02-03:
@jpliew did you have a chance to look into this? :)

jpliew, 2021-02-15:
@AuspeXeu sorry for the slow reply. I was stuck with work commitments. Please take a look at this and see if you can get it working

https://github.com/jpliew/Multi-Reader-Wiegand-Protocol-Library-for-Arduino

AuspeXeu, 2021-02-15:
I'll give it a shot an report back :)

AuspeXeu, 2021-02-26:
Works like a charm :)