monkeyboard/Wiegand-Protocol-Library-for-Arduino #53: Wiegand.h not working on other pins than 2 & 3

StephaneCYB, 2021-06-18:
Pin 2 is in our application used for the RS485 control, so I changed D0 to pin4 & D1 to pin5.

Any idea what I'm doing wrong?

#include <Wiegand.h>

WIEGAND wg;
const int WG_D0 = 4;
const int WG_D1 = 5;

void setup() {
  Serial.begin(9600);  
  
  // default Wiegand Pin 2 and Pin 3 see image on README.md
  // for non UNO board, use wg.begin(pinD0, pinD1) where pinD0 and pinD1 
  // are the pins connected to D0 and D1 of wiegand reader respectively.
  wg.begin(WG_D0,WG_D1);
}

void loop() {
  if(wg.available())
  {
    Serial.print("Wiegand HEX = ");
    Serial.print(wg.getCode(),HEX);
    Serial.print(", DECIMAL = ");
    Serial.print(wg.getCode());
    Serial.print(", Type W");
    Serial.println(wg.getWiegandType());    
  }
}

monkeyboard, 2021-06-19:
Arduino Nano that has ATmega328 only has two interrupt pins. This library
will not work.

You might want to consider this library that is using PinChangeInterrupt
https://github.com/jpliew/Multi-Reader-Wiegand-Protocol-Library-for-Arduino

Cheers

StephaneCYB, 2021-06-19:
Ok. Thank you !
Kind regards

closed 2021-06-19 by StephaneCYB