monkeyboard/Wiegand-Protocol-Library-for-Arduino #54: Order of bytes in "wg.getCode"

cedeps, 2021-08-02:
hello and first of all thank you for the library. 

I am new to arduino and programming, my question may seem naive to you.

I used the WiegandTest file from the Library. I want to retrieve part of the UID of a Mifare classic / ultralight tag.
I use this wiegand reader from BSTUOKEY which works in 5v: [https://fr.aliexpress.com/item/4000261733386.html?spm=a2g0o.store_pc_groupList.8148356.33.e079730aKqZWMp]

I understood that in wiegand 26 I was limited to the first 3 bytes
The code of the test file
`Serial.println (wg.getCode (), HEX` does display the start of the UID of my cards (Mifare classic or ultralight), but out of order:

For example for my card Mifare classic:
I have the result 9B3FAA while with another RFID reader and nfctools I have **AA**: **3F:** **9B**: 1A for a classic mifare.

For the ultralight:
68AF04 and with the other reader:
**04: AF: 68**: DA: C8: 5A: 80 (for example with my phone)

Why are the first 3 bytes inverted and is it possible to have them in the order of the UID?

Thank you in advance if you have a few minutes to inform me.
Cordially
ced

jpliew, 2021-08-02:
Hi,

The library is depending on the Wiegand reader to send the data, if the wiegand sends the data in reverse order, then you get a reverse order data.

The library will not know if the UID is valid or not. It is the reader's job to send to correct UID.

Hope this explains how it work.

If you are sure the bytes are reverse and still wish to use the reader, then you can reverse the data in your sketch.

Cheers.

cedeps, 2021-08-03:
Thank you very much for your answer.
I throw a bottle into the sea ...
Without wanting to waste your time, would you know how to get to reverse the result of (wg.getCode (), HEX)?
Not having any knowledge in this area, it is very complicated for me.
I searched with the leads you gave [https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/issues/37], but I can't do it. (wiegand 24) ... yet I searched!

Best regards

jpliew, 2021-08-04:
I would like to help you but when you said "but I can't do it" I am not sure what you meant. 

It will be easier to explain what you did, and what error you get, screenshot, errrors, etc. So that other people can understand your issue and offer you the right solution.

cedeps, 2021-08-05:
Thank you for your reply.
I looked for a Librarie that could do this reversal work. I turned to bithepers.h.
I had some problems with the serialPrint because the "0" disappeared if the code started with 0, for example 04AA56.

For more clarity, I send you the  code for my case, if it can be useful for others.

Arduino library
BitHelpers : [https://github.com/RobTillaart/bitHelpers](url)
Wiegand: [https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino](url)

Code 1: for WEMOS D1 mini ESP 32, reader 24 bits [https://fr.aliexpress.com/item/4000261733386.html?spm=a2g0o.store_pc_groupList.8148356.33.e079730aKqZWMp](url)

```
#include <bitHelpers.h>
#include <Wiegand.h>

WIEGAND wg;

void setup() {
  Serial.begin(115200);  
  
  // default Wiegand Pin 2 and Pin 3 see image on README.md
  // for non UNO board, use wg.begin(pinD0, pinD1) where pinD0 and pinD1 
  // are the pins connected to D0 and D1 of wiegand reader respectively.
  wg.begin(26,25);
}

void loop() {
  if (wg.available ())
  {
   uint32_t x = wg.getCode ();    // 24 bit fits in 32
    x = byteReverse (x);           // reverse 4 bytes
    x = x >> 8;                    // remove the 8 extra bits
    printHex( x, 6);
     }
}

void printHex(uint32_t x, uint8_t digits)
{
  uint32_t mask = 1ULL << (digits - 1) * 4;
  while (x < mask)
  {
    Serial.print('0');
    mask >>= 4;
  }
  Serial.print(x, HEX);
}
```
Thank you again for your help,
Ced

closed 2021-08-05 by cedeps