monkeyboard/Wiegand-Protocol-Library-for-Arduino #57: Possible get card code reverse with command

hinkma, 2022-01-05:
Have possible give somehow direct card code reverse example 9B3FAA, AA3F9B?
Need new part to software? Example wg.getCode (), HEX / wg.getCodeinv (), HEX
Some reader give data reverse.

jpliew, 2022-01-20:
Hey @hinkma 

There are C routines that does the reversal of bits, some pointers here https://www.google.com/search?client=firefox-b-d&q=c+programming+reverse+bit+order

mrNo0b, 2022-11-16:
I have the same problem...
Why it reads in reverse in the first place? what might cause this issue?
btw I also got 3 pairs of hex chars at first but when used NFC Tools on android to read the same tag I had 4 pairs, I have WG26/34 wire (gray) on my RFID reader so I connected it to ground and now I get full 4 pairs hex (only in reverse...)
I also tried to flip the D0,D1 wires and got completely different code.

Is it possible to add bool parameter to `void WIEGAND::begin(int pinD0, int pinD1)` that will reverse card code?

Is it possible to read data from tag?

And I think it would be good to turn on discussions for this repository.