monkeyboard/Wiegand-Protocol-Library-for-Arduino #59: Output bits are flipped.

Proggopogo, 2022-02-02:
Hello, I'm using a WG 34 reader, and ran into an issue. Reading the card with a standard reader drops these hex values: 4DE2D147, the library returns this output: HEX = D1E24D, DECIMAL = 13754957 As you can see the order is flipped, and two numbers are missing. How do I get these numbers, and how do I re-arrange the order that the bits are presented propperly?

jpliew, 2022-02-03:
Hi @Proggopogo could you try [Wiegand NG Library](https://github.com/jpliew/Wiegand-NG-Multi-Bit-Wiegand-Library-for-Arduino) and show me the output please?

Proggopogo, 2022-02-03:
Sure thing, give me a minute.

Proggopogo, 2022-02-03:
Alright, got the data. The Output is as follows:

Proggopogo, 2022-02-03:
Bits=34
RAW Binary=0000000001101000111100010010011011

The reader has fixed settings. Either it uses the WG26 protocol, or the WG34 one. Changing between the two shouldn't have much impact on the output, or should it?

jpliew, 2022-02-03:
```
Bits=34
RAW Binary     = 0000000001101000111100010010011011
Remove Parity  =  00000000110100011110001001001101

Convert to HEX after remove parity = D1E24D

Bits=26
RAW Binary    = 11101000111100010010011010
Remove Patiry =  110100011110001001001101

Convert to HEX after remove parity = D1E24D
```
Your Wiegand reader is producing consistent result for W34 and W26. The data is correct.

As far as this library is concern, the data produced by the Wiegand reader is read and printed correctly.  

Are you comparing the DATA printed on the CARD or DATA being read by another reader like a NFC mobile phone?

Proggopogo, 2022-02-03:
I just got the reader and the task to make the hex values comparable to that of an already installed one. The reader I need to make this work with is waterproof, hence the specification. The other reader, already in use, is probably an NFC one, this one right here to be precise:  [https://www.elatec-shop.de/twn3-modelle/twn3-multi-125](url)

jpliew, 2022-02-03:
In order to make them the same, you can just swap the bytes and ignore the last byte of the 4DE2D147 you got from the USB/RS232 reader.

4D E2 D1 D7 , drop last byte

4D E2 D1, swap first byte and last byte

D1 E2 4D = Wiegand Output

Proggopogo, 2022-02-09:
Yeah that'll do... The last bight might be a reading error on the USB one. I found an article about it, stating that most readers have a one byte userID and 4 byte Card ID. The Software that uses 4 Byte might just take the userID Byte and add it to the Card ID bytes.

closed 2022-02-10 by Proggopogo