monkeyboard/Wiegand-Protocol-Library-for-Arduino #60: Does the reader support 125Khz readers?

Proggopogo, 2022-02-16:
I have received a reader which is similar to the one we discussed over [here](https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/issues/59), but this one uses the 125khz standard instead of the Mhz standard. I receive no output whatsoever, even tho the wiring is the same, and it's all properly connected. Any ideas?

jpliew, 2022-02-16:
Hi @Proggopogo , this library basically read the Wiegand data and is not aware of what reader. Therefore as long as the reader is sending out Wiegand data. 125KHz or Mhz is not relevant.  Please see explanation below.

CARD----->(Khz/Mhz)---->READER----->Wiegand---->(This library)

Proggopogo, 2022-02-17:
Thanks a bunch! I had to add additional pull-up resistors to make the whole thing work.