monkeyboard/Wiegand-Protocol-Library-for-Arduino #61: Wiegand 26 to Nodemcu Amica

J0E45, 2022-06-13:
i got problem when using nodemcu for reading using RFID reader wg26. The problem is serial monitor wont show the id from my tag and always displays a question mark at serial monitor every time I press reset on nodemcu.

jpliew, 2022-06-14:
Will this help? https://forum.arduino.cc/t/why-am-i-getting-mirrored-question-mark-on-serial-monitor/692694

J0E45, 2022-06-14:
aw I'm sorry, I didn't define the GPIO pin to be the RFID reading data I/O pin. After I looked at AuspeXeu's program on the topic 'Connect two reader to one controller' I needed to define pins as in line.

this component does not work on my nodemcu when connected to the RX TX pin, but it works fine when the pin defined is different.

[image]