monkeyboard/Wiegand-Protocol-Library-for-Arduino #64: Is it possible to read card ID and memory?

mrNo0b, 2022-12-12:
I use this library, everything works fine, but I would also like to read data that I write to the cards/tags.
Is it even possible with WG?
Do I have to buy specific WG RFID reader or everyone can read memory?
I don't know much about Wiegand protocol and RFID that why I ask,,,

jpliew, 2022-12-12:
Hi @mrNo0b 

The Wiegand RFID reader will only output the card ID. 

To read data written into user area, you need RFID reader like this

13.56Mhz https://www.google.com/search?client=firefox-b-d&q=MFRC522

mrNo0b, 2022-12-12:
Thank you for the info!

I use this:
[image]

I also have MFRC522 but the problem is that this board is not exactly "production ready" I will need some case for it and it requires too many wires instead of D0 and D1...😢

closed 2022-12-12 by mrNo0b